ysoldak/HeadTracker #30: A noob question about battery

donne1226, 2023-05-24:
I wanna make it without the screen.
What is the hourly power consumption？ I wanna use it about 6 - 10 hours. Can you recommend what the battery capacity should be?

dlktdr, 2023-05-24:
This sounds very familiar to an issue I had with the our firmware in the past if the module you have in your radio isn't the PARA module. If you have the older non-para module it will connect but no data is sent. The older module required encryption to be enabled.

Yuri, here is all the info we found on that issue https://github.com/dlktdr/HeadTracker/issues/22 including some Bluetooth scans.

Although if this is the case then it probably never would have worked even once. Just a thought :)

donne1226, 2023-05-25:
Yes Cliff, I am using the Frsky non-PARA module.

I am sure it worked a few times because I can see value changes on the trainer page. When I did the range test, the radio played 'trainer signal lost' and 'trainer signal recovered'. by the way, I found that the connection range is very short and disconnects at less than one meter.

ysoldak, 2023-05-25:
> by the way, I found that the connection range is very short and disconnects at less than one meter.

I’d assume the board has real tiny antenna and not designed for longer distances.
In my experience the range was more like two meters or so. But then, I don’t see that as a problem, why would your goggles be more than a meter away from your radio :)

ysoldak, 2023-05-25:
Alright, so apparently there are more than one type of bluetooth module that can be installed in radios.

The correct one (works with this head tracker and equal to installed in recent FrSky radios) is called [PARA Wireless Module](https://www.horusrc.com/en/frsky-horus-x10-para-wireless-module.html).

And this is [WRONG module](https://www.horusrc.com/en/frsky-transmitter-horus-x12s-bluetooth-module.html), do NOT buy and install it, will not work, at least not with this head tracker.

donne1226, 2023-05-26:
Alright then.  Thanks anyway. Using such a small board for the head tracker is a great idea！

closed 2023-05-26 by donne1226

ysoldak, 2023-05-26:
You can still use XIAO with your radio, two options:
- use `PPM` (wired connection)
- use `Arduino Nano 33 BLE` on the radio side as a helper module (from Cliff's project), HT will connect to it wirelessly (same protocol) and helper module will connect to your radio via SBUS or PPM.

dlktdr, 2023-05-26:
@donne1226 You could also replace the module in your radio with an ESP board (Wroom32, pico32 or c3mini). You can flash some code that emulates the PARA module that should work with this Head Tracker too.

I was looking for a way to remove the dependency of using Frsky modules 😉 It's not super easy to remove the module after it's soldered in, so @ysoldak options are probably the better choice if your not up for a bit of a challenge. Code is here https://github.com/dlktdr/BTWifiModule

ysoldak, 2023-05-26:
Nice, @dlktdr !
As a XIAO fanboy, I want to hint on XIAO ESP32C3 board: https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html

dlktdr, 2023-05-26:
👍 Should (... hopefully) just have to change these pin numbers to make it work :) https://github.com/dlktdr/BTWifiModule/blob/245e0557da3e2e1c900752333933942caf901a6d/src/defines.h#L26C1-L27

I don't have one of these yet. Also like the size of the XIAO boards, only have the NRF versions currently. I might have to pick one up. I don't want to get too far off topic here, let me know if you ever give it a try 😉

donne1226, 2023-05-26:
I'm willing to tinker with the XIAO module to make it functional, but I don't think it's suitable for widespread promotion. First, it's a bit too expensive. Second, there are some technical difficulties.

In the past, I made a tutorial video based on Cliff’s project. Lots of people made their own head tracker modules, which was great!

I found this project a few days ago, and the XIAO board is smaller and cheaper than an Arduino. It's perfect for a standalone module with a built-in battery, so we can get rid of all wires. If you can make it work, I believe that more people will use it.

ysoldak, 2023-05-26:
Yes, I can try, have couple of such boards here and can test it by plugging in to PPM on my radio.

ysoldak, 2023-05-26:
FYI, gentlemen, to make information we discussed here more visible, I've created a [discussion with options for people w/o built-in PARA module](https://github.com/ysoldak/HeadTracker/discussions/31).