ysoldak/HeadTracker #35: After I flash the firmware to XIAO BLE Sense, can I connect directly to my transmitter's Bluetooth?

ZhouYixun, 2023-09-21:
After I flash the firmware to XIAO BLE Sense, can I connect directly to my transmitter's Bluetooth?

My situation is quite special. I connect directly to the transmitter of RadioMaster/BetaFPV via USB and send CRSF protocol to let the transmitter communicate with my rx. I do not use radio directly. So if I use a head tracker, can I use Bluetooth directly? What about connecting to my transmitter to control a specific channel?

ysoldak, 2023-09-21:
Hi, Eason!
Your setup is quite special indeed. And I have no experience with CRSF and its USB pass-through.

I can only answer about HT to Radio connection. So HT connects to radio via trainer (so-called PARA) protocol. Radio must have PARA compatible bluetooth module, see for more info: https://github.com/ysoldak/HeadTracker/discussions/31

From your radio point of view there is another (student) radio nearby and if you configure your mixers and outputs correctly, channels from student radio (Head Tracker in fact) are mapped to some channels your radio sends to your model.

So you can factor out HT actually from the equation and ask yourself, if there would be another radio connected via trainer connection (wireless or wired, no difference) to your radio, would your CRSF+USB setup be able to catch there TR1-3 input channels and add them to outputs?

Feel free to join our little FB group, find me there and we can discuss in messenger, probably easier.
https://www.facebook.com/groups/242161055041994

ZhouYixun, 2023-09-21:
how about Discord? It's a bit difficult to access facebook in my area

ZhouYixun, 2023-09-21:
Can HT connect to ELRS's Bluetooth via Bluetooth and then communicate directly?

ysoldak, 2023-09-22:
> Can HT connect to ELRS's Bluetooth via Bluetooth and then communicate directly?

This HT talks Bluetooth based PARA trainer protocol (FrSky radios, OpenTX radios).
It shall be possible to implement a feature so HT talks to ELRS Bluetooth. It is not implemented though, you are welcome to contribute :)

closed 2023-09-25 by ZhouYixun