ysoldak/HeadTracker #47: Adding Crossfire protocol?

ProfDrYoMan, 2024-04-15:
Hi,

adding crossfire would allow using ELRS as trainer link from head tracker to radio. 

Recent versions of ELRS allow reprogramming RX as TX, so a tiny RX converted to TX could be used on the head tracker. 

Edge-TX supports Crossfire as trainer input.

ysoldak, 2024-04-15:
I do not use Crossfire or ELRS myself, so chances are slim I ever implement this.
But I'm happy and open for contributions.

Just for me to have the full picture.
1) We are talking about radios w/o built-in bluetooth or with bluetooth but which does not speak PARA trainer protocol; otherwise no point to implement yet another link.
2) An ELRS receiver flashed and configured as transmitter (TX) will need to be connected to HeadTracker board (XIAO BLE Sense).

What's required from HeadTracker then? How shall it communicate with the ELRS TX board? Shall HeadTracker send data to serial port (RX/TX pins) in Crossfire format?

ProfDrYoMan, 2024-04-16:
Hi there,

1 Yes and no. I guess every TX that understands Crossfire will also understand para. Crossfire as input is only on EdgeTX afaik. So your reasoning is not wrong. Sometimes it is easier to fit an connect a small ELRS RX with ceramic antenna into the TX. 
I see that BT is already in head tracker side. So it’s straight forward using BT on TX side. 

2 The ELRS RX flashed as TX gets its data over serial in crossfire protocol. 

One other project uses this to transmit directly to the gimbal with the ELRS PPM RX directly at the gimbal. With all mixing in the head tracker.

ysoldak, 2024-05-02:
The other project being https://github.com/dlktdr/HeadTracker perhaps?

I want to keep this Head Tracker's code compact and zero-configuration, this means all mixing and scaling have to happen in radio.

ProfDrYoMan, 2024-05-02:
Fully understand. Closing this one.

closed 2024-05-02 by ProfDrYoMan