ysoldak/HeadTracker #48: permanent reset all 35 sec

boomerschnecke, 2024-04-19:
Having installed the latest SW to XIAO Sense board.
Using as Heattracker at Goggles.
Getting all 35sec periodically a reset of the board
Blue LED is switched off and Bluetooth modul at Radio side sets channels to 0 what means, that  Gimbal is resetted for a short time (around 1 sec)
XIOA is restarting and the Gimbal goes back to the needed position
Reproducable behaviour

ysoldak, 2024-04-22:
- Do you use version 2.5.1 from this repository?
- What is your radio transmitter?
- How do you connect head tracker to your radio?

boomerschnecke, 2024-04-22:
*yes, 2.5.1 latest version from your github page https://github.com/ysoldak/HeadTracker
*Radio = Radiomaster T16S
Seeduino XIAO BLE Sense with SW from https://headtracker.gitbook.io/head-tracker-v2.2 via Bluetooth and UART - Trainer Port conncetion

ysoldak, 2024-04-22:
So tell me is this ALL correct?
- On your goggles you have XIAO BLE Sense flashed with this repo FW 2.5.1;
- On your T16S radio you have another XIAO BLE Sense connected to the radio via UART trainer port, this XIAO has Cliff's firmware flashed and configured to run in "helper" or "rx" mode;
- Both XIAO boards do connect and talk via Bluetooth, but connection drops periodically (you say every 35 secs)

ALL this above true?
I, honestly, never tested such setup myself. All my radios have BT module built-in, so 🤷 

Does the radio say "trainer lost" and "trainer recovered" when the disconnect happens?

boomerschnecke, 2024-04-29:
I did answer to you last week, but it seems something has gone wrong:
1) yes at XIAO BLE Sense there was the FW 2.5.1 installed (in the meantime I tried to use also the FW from Cliff)
2) yes, at T16s XIAO BLE Sense with SW from Cliff, connected via Bluetooth to XIAO BLE Sense at Goggles2. This one is connected via UARTexternal (AUX1) to the T16S, which is for this two channels in Trainer Mode setup
Thats true of course, otherwise I would not waste my time for giving you this information. I thought this might be interresting for you. From T16s there was coming no information concerning lost trainer of something like this. "Only" a "reset" to 0 for the servos all 35sec

ysoldak, 2024-05-02:
I would think this is the new feature of re-center on double tap misbehaving. Added in 2.5.0
But you say it does this weird stuff every 35 sec, that's really odd.

If you can, try connect a serial monitor (say from Arduino IDE) to the XIAO BLE board on goggles and see what's the output. Also or instead you may try and connect 128x32 little display to the board and see if axes jump there too.

I'm trying to understand where problem is, in this head tracker or in the helper module connected to your radio or somewhere in-between.

boomerschnecke, 2024-05-02:
Sorry, I do not have any of the mentioned additonal equipment available

Thetruesaint, 2024-06-22:
Hi guys!, me and a friend have a similar problem:
Seeduino XIAO BLE Sense with SW 2.5.1
Radio: TX16S MRKII with an ESP32 C3 supermini with proyect: https://github.com/dlktdr/BTWifiModule.git

Exactly 30 sec. after conection, it disconect from radio (Trainer signal Lost) and we see blue led flash again in the Xiao. In my case, some times, it`s reconnect inmidiatly... in my friend case, it always reconnect every 30 sec. 

On the serial monitor, we can see axes movement, and no change in the output at the disconection event (86 to 87)

[image]

We still have to try to exchange Xiaos and see if the problem move or stay with the radio.

ysoldak, 2024-06-23:
@Thetruesaint thanks for report, see similar #52