ysoldak/HeadTracker #49: Is a base for DJI Goggles 2 existing

boomerschnecke, 2024-04-19:
Hi looking for a XIAO Base to be used for my DJI Goggles 2.
Or could I use one of the DJI Integra XIAO BLE Base for Goggles 2?
Could you please provide some detailed photos how this should be mounted and fitted?
Do I have to modify the foam? Or is it possible to use without faom modification?
Thank you

ysoldak, 2024-04-19:
Please see here and respective STL file

[image]

ysoldak, 2024-04-19:
DJI Integra and goggles 2 is the same. No foam modification needed.

boomerschnecke, 2024-04-19:
Thanks a lot for your quick reply. The issue is, that Goggles 2 is not having this "handle bar" at the top of the Goggles. So there is no possibility to click this "bar" in. Thats why a ask if something additional specially for Goggles 2 is existing. Or do I have to glue it on the foam?
https://store.dji.com/de/product/dji-goggles-2?vid=127611&from=search-result-v2&position=1&total_result=46
https://store.dji.com/de/product/dji-goggles-integra?vid=135261&from=search-result-v2&position=0&total_result=13
Have a look to the top in the middle of goggles

ysoldak, 2024-04-22:
Oh, my bad, I have Integra only and for some reason assumed Goggles 2 have top strap also.
Well, no then, nothing specific exists for Googles 2 yet. Will write here if/when we have a solution.