ysoldak/HeadTracker #50: PPM not working?

ysoldak, 2024-05-02:
Assembling here bug reports from @KPOPRC in no particular order.
Let's track the issue here.

>yuri! great project! did you stop supporting PWM on all the new releases. i can't get any TR signals via pwm from any of your releases. am i doing something wrong?

>Hello! it's looking like a great project. but the only stable release i can find that doesnt crash with in 3 sec, is 2.3.4 with pwm pin grounded. None of the other versions have red and green flashing with solid blue. regardless no head tracking values are making it to any of the TR channels. Did you guys stop supporting pwm? Please help. TR4 the channel goes up to -90%. all other tr channels kind of glitch at -7. when i unplug the trainer cable, i get an audible "trainer single lost". Any version after 2.3.4, no trainer signal is output via pin d10. D8 is grounded.

>Hello! After version 2.4 pwm doesn't recognize via my radio. But all UF2 files have never worked all the way down to 2.5.0. None of the TR channels carry any head tracking values. After 2.4 the board only flashes red. Can you help with this? I'd love to get these 2 head trackers working with pwm as my radiomaster doesn't have bluetooth.

ysoldak, 2024-05-02:
In your messages you probably mean PPM when you write PWM.

Please make sure you ordered correct XIAO board. You want XIAO BLE Sense variant, i.e. the one with IMU.
The cheaper board that is called simply XIAO BLE, without "Sense" in name, it will not work.

KPOPRC, 2024-05-02:
hey. thanks for the reply. this is the sense varient. I believe.

ysoldak, 2024-05-02:
The issue turned out to be the wrong board indeed. For the head tracker to work, it needs IMU (Inertial Measurement Unit) that is only available on XIAO BLE Sense board. See that little black box down left.

Closing the issue.

[image]

closed 2024-05-02 by ysoldak

KPOPRC, 2024-05-02:
thanks for the help. I did notice that even with out the imu. I was still
getting a trainer signal. but for the latest releases no tr signal was out
put. I assume this is still a 'wrong board problem'. I'll let you know how
it goes with the new boards. they'll be here tomorrow. thanks for your
time.

k