ysoldak/HeadTracker #51: Two Seeeduino XIAO BLE Sense

bodychmarcin, 2024-06-19:
Hi. Is it possible to connect two modules together? one as a transmitter and the other as a receiver installed in TARANIS which sends the received data to the PPM pin?

ysoldak, 2024-06-23:
Not at the moment, not for two XIAO BLE Sense boards.
Some however try to use ESP32 C3 board on the radio side, and have issues unfortunately, see #48 and #52

I'll try and implement receiver on XIAO BLE board, eventually. No time frame though.

bodychmarcin, 2024-06-23:
Lack of time is very understandable for me :-) generally it doesn't have to be the same board, it can be anything else like esp32.  Tracker 2.2 as a receiver was mentioned in some threads, but as a BLE receiver for PPM it is too advanced and I had a problem with it.  I used ESP32 to connect to your firmware with the Xiao board, the connection was not broken even for a moment, and it was running for over an hour.  The problem is that I don't understand and I don't know the uuid structure and data, and therefore I don't know "golang".  That's why I started this thread.  Best regards, my friend.

bodychmarcin, 2024-06-28:
I wrote a program based on opentxbt that connects to seeduino, it works fine, nothing interrupts the connection, I receive data and send ppm.
Question: I'm curious, is it possible to implement client subscription in seeduino?

ESP32 BT Client, GPIO short to Ground -> sends bool (true or false)
XIAO BT server, receives data bool, if true -> Reset orientation

It is more convenient to do this with a monostable rc switch than to take your hand off to tap twice or look for a button soldered to the XIAO GPIO.

RDM3DGit, 2024-06-28:
> I wrote a program based on opentxbt that connects to seeduino, it works fine, nothing interrupts the connection, I receive data and send ppm. Question: I'm curious, is it possible to implement [...]

Can you share link to your code?

bodychmarcin, 2024-07-01:
> Can you share link to your code?

I think so, but I would like to wait until ysoldak could create an additional characteristic, e.g. fff7, in which the client can save bool data, if xiao receives true, it will reset the orientation.

ysoldak, 2024-07-01:
Yes, good idea, @bodychmarcin!
Definitely we can have such extension to BT, add a characteristic. This will not be a strict PARA anymore, no problem though, since backwards compatible still.
Any known implementation/characteristic number for reset orientation? So we don't invent a wheel but just follow them?

bodychmarcin, 2024-07-01:
> Tak, dobry pomysł,@bodychmarcin! Zdecydowanie możemy mieć takie rozszerzenie do BT, dodać cechę. Nie będzie to już ścisła PARA, nie ma jednak problemu, ponieważ nadal jest kompatybilny wstecz. Jakiś znany numer implementacji/charakterystyki dla orientacji resetowania? Czyli nie wymyślamy koła, tylko po prostu za nimi podążamy?

I don't know the language, so I have to ask you about it.  I think you can add another number fff7.  What do you think about it?  unless it's a feature outside the Para structure.  I have ESP32C3 and the monostable switch is physically disconnected from the device and connected to the ESP32 GPIO.  if gpio is shorted, I would like it to send a bool to the xiao feature and reset it.  I don't want to take my hands off the radio :-)

ysoldak, 2024-07-01:
Yes, I think I understood you idea, adding one more characteristic shall be no problem.

So, the BT client sets a characteristic value to "true"; HT notices the value and resets its orientation + resets the characteristic value to default "false".

Shall work, need to test, obviously.

bodychmarcin, 2024-07-01:
I'll test it if you make a tag related to it :)

bodychmarcin, 2024-07-02:
I already managed to create a new service with new characteristics to handle the reset so as not to change PARA.  anyway, thanks for your willingness

closed 2024-07-02 by bodychmarcin