ysoldak/HeadTracker #52: Continuosly lost and regain trainer conection (BT)

RDM3DGit, 2024-06-23:
Radio type: TX16S MK2 JB edition
Versions used:
- Radio FW: Edgetx 2.10.1 centurion cloud builded with bluetooth flag enabled
- ESP32 C3 Super Mini (Modified definition.h C3Mini URTS pins): cloned github repository, builded it and uploaded via VSCode
Connecting to: Xiao BLE sense HT with 2.5.1 FW

Expected beheavour: Conection between devices stays solid and uninterrupted.

Actual beheavour: Conection stablished perfctly, but after a minute or so the conection is lost and recovered instantanously. This continues to happen every 30 o 40 seconds continusly.

RDM3DGit, 2024-06-23:
C3 super mini terminal output during disconections and reconection:

BTCLIENT: scan start success
Disc BT Address 032BEC51FFBF, RSSI=-75, Addr Type=1
Disc BT Address F66A731161C3, RSSI=-80, Addr Type=1
Disc BT Address 1C72AA9511E4, RSSI=-80, Addr Type=1
Disc BT Address F66A731161C3, RSSI=-84, Addr Type=1
Connecting to F66A731161C3
Unable to connect to F66A731161C3, address not found in storage
I (716620) BTCLIENT: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (716620) BTCLIENT: REMOTE BDA:
I (716620) BTCLIENT: f6 6a 73 11 61 c3
I (716620) BTCLIENT: Starting Service Scan
I (716630) BTCLIENT: open success
I (716910) BTCLIENT: update connection params status = 0, min_int = 10, max_int = 10,conn_int = 10,latency = 0, timeout = 70
I (717340) BTCLIENT: discover service complete conn_id 0
I (717340) BTCLIENT: SEARCH RES: conn_id = 0 is primary service 1
I (717340) BTCLIENT: start handle 23 end handle 65535 current handle value 23
I (717350) BTCLIENT: service found
I (717350) BTCLIENT: UUID16: fff0
I (717360) BTCLIENT: Get service information from remote device
I (717370) BTCLIENT: ESP_GATTC_SEARCH_CMPL_EVT
I (717370) BTCLIENT: Attr Count 5
I (717370) BTCLIENT: Found the Trainer Characteristic
I (717380) BTCLIENT: Subscribing for Notifications
I (717390) SETTINGS: Reading settings
I (717390) SETTINGS: Settings Read Successfully
I (717400) SETTINGS: No data differs, not writing to flash
I (717400) BTCLIENT: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0
I (717410) BTCLIENT: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (717450) BTCLIENT: write descr success
W (747450) BT_GATT: gatt_rsp_timeout disconnecting...
W (747460) BT_HCI: hci cmd send: disconnect: hdl 0x1, rsn:0x13
W (747460) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
E (747460) BTCLIENT: write char failed, error status = 85
W (747470) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x16
I (747470) BTCLIENT: ESP_GATTC_DISCONNECT_EVT, reason = 22
Connecting to F66A731161C3
Unable to connect to F66A731161C3, address not found in storage
W (747970) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (748270) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (748740) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
W (749220) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e
W (749220) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
I (749220) BTCLIENT: ESP_GATTC_DISCONNECT_EVT, reason = 62
E (749220) BTCLIENT: open failed, status 133

ysoldak, 2024-06-23:
Thank you @RDM3DGit for the great ticket!
I think it's the same as reported in #48 (or at least commented in)

TBH, I never did integration test with ESP32 C3 master, probably I should since I see more and more people use such combination.

ysoldak, 2024-06-23:
@RDM3DGit it looks like the problem is on ESP32 side though. BT stack seem to be misconfigured, etc.
See discussion here: https://devzone.nordicsemi.com/f/nordic-q-a/96053/esp32-gateway-disconnects-from-end-devices-nordic-after-30-seconds

I'm going to implement companion/receiver firmware for XIAO BLE Sense HT, on XIAO BLE [Sense] boards at first, then maybe support for ESP32 C3 comes too, but no promises. See #51

RDM3DGit, 2024-06-23:
Thank you very VERY much!!! for you quick response!

We are a group of RC FPV enthusiast here in Argetina looking foward to your HT project. Unfortunatly at this side of the world, it is very difficult and expensive for us to buy or even rich to some one that sells Arduino BLE 33 boards, thats way we are dealing with the hardware we could get here. (xiao ble sense on the HT side, and esp32 c3 super mini on the radio side).

I will read the discussion you pointed at, and keep trying.

@Thetruesaint FYI!

RDM3DGit, 2024-06-30:
Finally I go it to work without disconnections!!!! Im actually now connected since 40 minutes ago!!! The problem was at line 338 and 393 of bt_client.c of Cliff code where it was defined that gatt write requiered a response I changed them to not wait for a response and problem solved!!!
[image]

closed 2024-06-30 by RDM3DGit

ysoldak, 2024-07-01:
Great! Thanks for sharing!
I trust you tell Cliff about this.

At this point, I’m not sure yet is this bug in Cliff’s code or mine… In other words, did you just hack Cliff’s code to work with my HT (so I’d better fix on my side, so no hacks needed) OR it was indeed a bug and shall be fixed there (your change applied for everyone).

RDM3DGit, 2024-07-01:
I think it's a bug on cliffs code. There's no reason for the client to wait for a response there. 

Any way, I told him about this at his repository (closed the issue there too) but I don't think he is working on this code any more... (last update 2 years ago).

Thanks again, really love your code. I'm already using it. 
[image]