ysoldak/HeadTracker #53: Recenter Button Not working with BLE Arduino on Remote Side

Velvethsaw, 2024-08-09:
So I we able to get a the Seeed Xiao sense working as a blue tooth head tracker withing minutes using this program. I think I am getting much less drift from my orginal setup. This is a wonderful solution, thank you so much!👌

I was previously using two Arduino BLE sense and Cliffs headtracker version. I was able to substitute the Seeed Xiao on the head tracker side and everything is working BUT the physcial centering switch I have wired to the remote side Arduino BLE is no longer working. 

Anyway to fix this?

ysoldak, 2024-08-09:
Thank you, glad you like it!

I'll need to check how they send re-center signal back to HT unit, hopefully just setting a bluetooth attribute. If so, it will be real easy to implement. Just need to find time for that. Contributions are welcome!

ysoldak, 2024-08-09:
A hint: I know this is not something you used to and may not want to take your hand off your radio, but this HT can re-center w/o physical button -- just double-tap your goggles anywhere, and that shall re-center.

Velvethsaw, 2024-08-12:
Luckily I haven't not needed the recentering as much as before due to drift reduce on the Xaio board. Huge improvement. Unfortunately the system likes to drift at the most critical times during the flight so I need to keep my hands on the remote. I can lock the PWM positions with the flight control through my normal controls, then double tap my googles when I have a moment. But I would much rather have to option to have the additional button  to recenter right at my finger tops.

Thanks for the quick response. Bravo on the project!

dlktdr, 2024-08-19:
If you want to remote reset a 'R' character is written to Characteristic 0xAFF2,
https://github.com/dlktdr/HeadTracker/blob/6abe38db1ec6e71d9ade9d227abe93396ed7cb0b/firmware/src/src/btparahead.cpp#L323-L325