ysoldak/HeadTracker #54: Add a magnetometer

Doncuppjr, 2024-09-30:
I think the seeed would support it, but could the firmware be updated to support a magnetometer for better pan support? I think we would also need some kind of configuration options for defining which axis takes that input, as currently the tracker can be put into almost any orientation and depending on where you mount it, an axis represents the pan.

ysoldak, 2024-09-30:
Do you have a link to a Seeed board with magnetometer?

Yes, this is by design -- you put the tracker on your googles any way you see best and then you map in your radio the channels accordingly. No need to configure also in HT.

Doncuppjr, 2024-09-30:
Maybe as both the seeed and mag are 3 axis, their could be a pairing pattern that would mitigate the need for configuration.

ysoldak, 2024-10-01:
Sure, this is doable and we can define how a magnetometer board shall be mounted relative to main xiao board so we can skip configuration.

The usefulness though escapes me, since:

>For a long time, Nano 33 BLE was the only board supported. It has an additional sensor, magnetometer, that head trackers usually use to eliminate pan drift. In practice, however, magnetometer adds more problems than solves. Magnetometer is very sensitive to environment and tricky to calibrate properly. In this project we do not use magnetometer and have a good zero-configuration automatic continuous gyro calibration instead to solve the drift problem. Already after first 5 seconds the calibration is good enough to stop the drift.

Can be a great fun addition. But only if we can make magnetometer auto-calibrate (I have some [experiments in that direction](https://github.com/ysoldak/magcal)). It's doable also, but as said probably is not worth the trouble.

Do you have problems currently with HT drifting on Xiao boards?

Doncuppjr, 2024-10-01:
I don't know if I actually have drift problems. I'm just getting started with the HT, and some things don't work quite like I expect. I might try playing with the weights and see if that resolves what I'm experiencing, but it seems like drift might also be a factor. Maybe the HT is detecting a double tap and resetting or something.

I do wonder how important calibration actually is for this though. I don't expect the HT would actually need to output the heading to anything, but rather just a reference to the magnetic field to avoid any potential drift of the accelerometer. 

Can be a great fun addition. But only if we can make magnetometer auto-calibrate (I have some experiments in that direction). It's doable also, but as said probably is not worth the trouble.

Do you have problems currently with HT drifting on Xiao boards?

Doncuppjr, 2024-10-13:
So I can confirm that I have a drift issue. I have the tracker inputs on a switch. I can sit down, double tap and everything is fine. Fly around a bit, hit the switch and my gimbal moves about 60 degrees to the left. Release switch, double tap, hit the switch, and everything is fine. Release switch, fly around a bit, hit the switch and the gimbal moves to the left again.

ysoldak, 2025-02-01:
@Doncuppjr I've spent some time in lab these days and successfully reproduced the drift behaviour.

Drift happens after fast movements. No drift when stationary. 

I usually move my head around very slowly when flying, so did not notice the behaviour myself.
However, I was able to reliably reproduce the behaviour in my lab (board attached to a servo with servo tester and jerking it back and forth).

Full-fledged 9DOF IMU (with magnetometer) support may be required after all to fully eliminate the drift.

Meanwhile, I've played with sensor fusion parameters (see #57) and it seems like the HT handles fast movements much better now.

I'm going to release `v2.5.2` so you and anyone else interested can give it a try.

Leaving the ticket open until magnetometer support is finally added.

Doncuppjr, 2025-02-18:
Nice, I played around with wiring up a QMC5883 to the Xiao Sense. A basic arduino sketch shows that it works, but still some more work to do to make it a viable option.

[image]

```#include "LSM6DS3.h"
#include "Wire.h"
#include "QMC5883LCompass.h"

// Create an instance of class LSM6DS3
LSM6DS3 myIMU(I2C_MODE, 0x6A);  // I2C device address 0x6A

// Create an instance of the QMC5883LCompass class
QMC5883LCompass compass;

void setup() {
    Serial.begin(115200);
    Wire.begin();

    // Initialize LSM6DS3
    if (myIMU.begin() != 0) {
        Serial.println("LSM6DS3 Device error");
    } else {
        Serial.println("LSM6DS3 Device OK!");
    }

    // Initialize QMC5883
    compass.init();
    compass.setSmoothing(10, true);
    compass.setCalibration(-1150, 1250, -700, 1100, -750, 850);  // Adjust based on your calibration data

    Serial.println("QMC5883 initialized");
}

void loop() {
    // Accelerometer
    Serial.print("\nAccelerometer:\n");
    Serial.print(" X1 = ");
    Serial.println(myIMU.readFloatAccelX(), 4);
    Serial.print(" Y1 = ");
    Serial.println(myIMU.readFloatAccelY(), 4);
    Serial.print(" Z1 = ");
    Serial.println(myIMU.readFloatAccelZ(), 4);

    // Gyroscope
    Serial.print("\nGyroscope:\n");
    Serial.print(" X1 = ");
    Serial.println(myIMU.readFloatGyroX(), 4);
    Serial.print(" Y1 = ");
    Serial.println(myIMU.readFloatGyroY(), 4);
    Serial.print(" Z1 = ");
    Serial.println(myIMU.readFloatGyroZ(), 4);

    // Thermometer
    Serial.print("\nThermometer:\n");
    Serial.print(" Degrees C1 = ");
    Serial.println(myIMU.readTempC(), 4);
    Serial.print(" Degrees F1 = ");
    Serial.println(myIMU.readTempF(), 4);

    // Magnetometer
    Serial.print("\nMagnetometer:\n");
    compass.read();
    Serial.print(" X2 = ");
    Serial.println(compass.getX(), 4);
    Serial.print(" Y2 = ");
    Serial.println(compass.getY(), 4);
    Serial.print(" Z2 = ");
    Serial.println(compass.getZ(), 4);

    delay(1000);
}```