ysoldak/HeadTracker #56: How to add different or more inputs

Podex12, 2025-01-15:
First of all, I'm sorry if I'm being disrespectful of this following question/request. I have been wanting to make an additional input device for my ground station. I'm thinking of making a device consisting of multiple 2 or 3 position switch and a stick connected to esp32 (i have a lot of unused esp32 and esp8266) with its own power supply (probably 1s lipo). I plan to use the stick to control a camera gimbal pan and tilt rather than a magnetometer, while the switches is used for lights, auxiliary, autopilot/mission mode, etc. The problem is i don't know how could i send those inputs to my TX, mainly the Sofware side. My first guess is using the trainer port. So if you don't mind can you tell me how could i do this?
Thank you in advance

ysoldak, 2025-01-15:
Hi, @Podex12 
Totally fine question, no worries!
So the simplest (single?) way to send data to you radio from an external device is via trainer port.
Most of radios have a physical trainer port (headphones jack) that accepts channels data encoded as PPM.
Some radios have Bluetooth built-in, that talks PARA wireless trainer protocol -- one can connect wirelessly to them (convenient) and send channel data that way.

We can talk in more details, feel free to reach out to me, search for Yurii Soldak on FB or just on internet and drop me a message!

ysoldak, 2025-01-16:
Nothing to do, this was a request for clarification.

closed 2025-01-16 by ysoldak